cherts/esp8266-devkit #2: Error with new version

fvpalha, 2015-04-29:
Hi Mikhail.

I removed completely the old version 2.0.2 and I installed the new version 2.0.3, but now my project doesn't compile.

```
**** Build of configuration Default for project ESP8266-transparent-bridge-master ****
mingw32-make.exe -f C:/Espressif/examples/ESP8266-transparent-bridge-master/Makefile all 
CC driver/gpio16.c
CC driver/uart.c
CC user/config.c
CC user/flash_param.c
CC user/server.c
CC user/user_main.c
AR build/app_app.a
LD build/app.out
c:/espressif/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: build/app.out section `.text' will not fit in region `iram1_0_seg'
collect2.exe: error: ld returned 1 exit status
mingw32-make.exe: *** [build/app.out] Error 1
```
The same occurs with the example ESPWebFramework.

```
**** Build of configuration Default for project ESPWebFramework ****
mingw32-make.exe -f C:/Espressif/examples/ESPWebFramework/Makefile all 
C+ user/fdvcollections.cpp
C+ user/fdvconfmanager.cpp
C+ user/fdvdebug.cpp
C+ user/fdvflash.cpp
C+ user/fdvnetwork.cpp
C+ user/fdvprintf.cpp
C+ user/fdvserial.cpp
C+ user/fdvstrings.cpp
C+ user/fdvsync.cpp
C+ user/fdvutils.cpp
C+ user/user_main.cpp
AR build/app_app.a
LD build/app.out
c:/espressif/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: build/app.out section `.text' will not fit in region `iram1_0_seg'
collect2.exe: error: ld returned 1 exit status
mingw32-make.exe: *** [build/app.out] Error 1
```
Do you have a tip?
Best regards.

CHERTS, 2015-05-01:
Yes, there is such a problem, I'll try to fix it soon

CHERTS, 2015-05-09:
This bug has been fixed in version DevKit 2.0.4, you can download it [here](https://github.com/CHERTS/esp8266-devkit/releases/tag/v2.0.4)

closed 2015-05-09 by CHERTS

fvpalha, 2015-05-11:
Thank you. This project is wonderfull, congratulations Mikhail.