cherts/esp8266-devkit #3: I got compiler Error in Espressif-ESP8266-DevKit-v2.0.4-x86 (build/app.out section `.text' will not fit in region `iram1_0_seg')

Smith1966, 2015-05-22:
Hi, please help to solve this error "build/app.out section `.text' will not fit in region `iram1_0_seg'".

CHERTS, 2015-05-22:
What an example of such a mistake?

Smith1966, 2015-05-23:
I was trying to modify an eample to use FatFs files (http://elm-chan.org/fsw/ff/ff11.zip Generic)  but it happened and I can not use it but with bigger and another source code it is okay.

fvpalha, 2015-05-28:
Hi @Smith1966.
The project [FatFs - Generic FAT File System Module](http://elm-chan.org/fsw/ff/00index_e.html) is really interesting.
Maybe the [ChaN](mailto:user1@elm-chan.org) can help you.
Please, keep us updated.

MrZANE42, 2015-06-14:
Unless the makefile is constructed so that code is  NOT cached to RAM at boot time you need to add the attribute ICACHE_FLASH_ATTR on every function. See how it's done in the examples. Good luck

closed 2015-06-14 by CHERTS