cherts/esp8266-devkit #5: when release with SDK 1.1

tinamore, 2015-05-28:
Hi
When you release with SDK 1.1 ?
Please release soon
Thanks

fvpalha, 2015-05-28:
Hi.
Warning: Patch based on SDK_v1.1.0 - http://bbs.espressif.com/viewtopic.php?f=5&t=508

Thanks.

tinamore, 2015-05-28:
Hi
I have a question:

Does "https://github.com/CHERTS/esp8266-devkit/tree/master/Espressif/ESP8266_SDK" have include SDK_v1.1.0 from "http://bbs.espressif.com/viewtopic.php?f=5&t=508" ?

I have zip "https://github.com/CHERTS/esp8266-devkit/tree/master/Espressif/ESP8266_SDK" and replace in old version C:\Espressif\ESP8266_SDK at my computer

Do I need to Patch based on SDK_v 1.1.0-http://bbs.espressif.com/viewtopic.php?f=5&t=508 ?

Thanks

CHERTS, 2015-05-28:
SDK v1.1.0 is available in the repository, you can download the current version [here](https://github.com/CHERTS/esp8266-devkit/archive/master.zip)

Delete the directory C:\Espressif\ESP8266_SDK and C:\Espressif\examples, and copy the new directories ESP8266_SDK, ESP8266_SDK_101 and examples from the archives, a link to the file above.

tinamore, 2015-05-28:
Ok, thank you.
Great project :)

closed 2015-05-28 by tinamore