cherts/esp8266-devkit #11: Custom installation path / Drive Selection option

ganeshkumarsw, 2015-08-13:
Custom Installation path option shall be available for Espressif-ESP8266-DevKit windows application. It normally installs in C drive but it shall be configurable.

CHERTS, 2015-08-14:
You can copy the folder C:\Espressif to another drive, but then you will need to change in the python-files (gen_appbin.py and others) from the folder C:\Espressif\utils way to the gcc compiler and compile a new executable files (*. exe). To compile the python-to-exe to install Python 2.7

CHERTS, 2015-08-14:
Also, you will need to change paths in project files Makefile and Eclipse

ganeshkumarsw, 2015-08-14:
Can this drive option change be available as a part of installer?

CHERTS, 2015-08-14:
Yes, I plan to do the installation in any location