cherts/esp8266-devkit #13: ESP8266_RELAY_BOARD project stacked on restarting

majercakdavid, 2015-08-20:
I wanted to try an ESP8266_RELAY_BOARD example but after flashing it and trying to boot it the device just stacks on restarting. Do you have any solution ?

Def152, 2015-08-28:
+1

I'm using CHERTS devkit with sdk v1.3.0. Everything is compiling and flashing fine. But every 3 seconds esp is resetting with boot message (3, 7). I assumed it's power supply fault. I made another power supply 1A, 3,3V but this didn't helped. I also tried esp arduino and it's working like charm. So I thought it might be makefile. I'm using esp-12e with 32Mb, this is my makefile http://pastebin.com/p28REHFC for blink example

CHERTS, 2015-09-01:
The [DevKit version 2.0.8](https://github.com/CHERTS/esp8266-devkit/releases/tag/v2.0.8) example ESP8266_Relay_Board now works correctly.

closed 2015-09-01 by CHERTS