cherts/esp8266-devkit #17: Add the rboot sample in package

fvpalha, 2015-09-28:
Hi Mikhail,

I suggest add the FOTA example with [rboot](https://github.com/raburton/esp8266).

I use the [UDK[(https://github.com/CHERTS/esp8266-devkit) in our project and I am having trouble with the Makefile.

Kind regards.

fvpalha, 2015-09-28:
Hi Mikhail,

I commited the sample in my fork [esp8266-devkit](https://github.com/fvpalha/esp8266-devkit/commit/832db9e4049766a216e30c44321721e87737c109).

Kind regards.

CHERTS, 2015-09-29:
Hi
a very good example, I am ready to accept in devkit, please, create a pull
requests.

-- 
--------------------------------------------------------------------------
С наилучшими пожеланиями, Михаил Григорьев
Руководитель группы разработчиков
--------------------------------------------------------------------------
Россия
г. Челябинск
WWW: www.novell.chel.ru
WWW: www.im-history.ru
WWW: www.programs74.ru
E-Mail: sleuthhound@gmail.com
ICQ: 161867489
--------------------------------------------------------------------------
With the best regards, Mikhail Grigorev
Lead developer
--------------------------------------------------------------------------
Russia
Chelyabinsk
WWW: www.novell.chel.ru
WWW: www.im-history.ru
WWW: www.programs74.ru
E-Mail: sleuthhound@gmail.com
ICQ: 161867489
--------------------------------------------------------------------------

fvpalha, 2015-09-29:
Hi Mikhail,

I created the [pull request](https://github.com/CHERTS/esp8266-devkit/pull/18).

If possible, I would like your help to change the Makefile.

CHERTS, 2015-10-02:
Thank Fernando, good work.

closed 2015-10-02 by CHERTS