cherts/esp8266-devkit #28: ESP32 Toolchain support

hreintke, 2015-12-23:
Hi,
I am active in the Sming framework and there is currently work in progress to include ESP32 support in there.
For ESP8266 the windows users of sming are using Cherts UDK as a base for their development environment.

Are there any plans for adding the ESP32 toolchain for windows to the UDK releases ?
If so, can you indicate the progress you have in that and, if possible, any timelines on that ?

robotiko, 2015-12-23:
+10

CHERTS, 2015-12-24:
Development Unofficial DevKit for ESP 32 is almost finished, you can try the first [beta-version](http://programs74.ru/get.php?file=EspressifESP32DevKit).
It can be set and used in conjunction with Unofficial DevKit ESP8266.
While there is only one example - esp32_hello_world
I can not check to stitch this example, I have no board ESP32.

flexiti, 2015-12-24:
and ESP32 is somewhere to buy?

hreintke, 2015-12-24:
Thanks,
I took the beta version and successfully compiled the hello_world example.
Appreciate your work on this, will use this for my ESP32 Sming support and feedback results.

CHERTS, 2015-12-24:
>>and ESP32 is somewhere to buy?
I do not know where to buy a ESP 32, it has not yet sold. I do not have it :(

closed 2016-02-04 by CHERTS