cherts/esp8266-devkit #30: SSL Client

Draco785, 2016-03-01:
Hi,

I was able to install with success this DevKit and run several of the demos on a NodeMCU 0.9.

Currently, I'm trying to establish a secure SSL connection, based on the code published here: [http://bbs.espressif.com/viewtopic.php?f=31&t=389&p=1466&hilit=ssl#p1466]

After adding the corresponding dependencies, like `#include "espconn.h"`, and others, I'm receiving the errors:
>undefined reference to `espconn_secure_connect'
>undefined reference to `espconn_secure_sent'

Is it anything wrong with that code? Does the secure-connection part of the SDK work OK? Do you have plans for an example of a secure connection?

Thanks for the help.

icodk, 2016-03-01:
Do you have this line in your Makefile-uer.mk ?
EXTRA_LIBS= ssl crypto

Draco785, 2016-03-01:
I don't have a .mk file in the project.
The current Makefile was copied from the HelloWorld.
Could you give me the instructions for working with it?
Thanks.

Draco785, 2016-03-01:
Well, I added ssl to the flag LIBS in the Makefile, and the error is gone.
Thanks for the hint.

closed 2016-03-01 by Draco785