cherts/esp8266-devkit #31: Ripo updated with SDK v1.5.2, installer is still old version

chall3ng3r, 2016-04-25:
There are updates in this ripo about 2 months ago, which include patches to compilers and SDK update to 1.5.2. But the pre-built installer available on linked website is still older one.

I'd appreciate if maintainer also update the installer with these latest changes. Thanks.

CHERTS, 2016-04-26:
You can use the version from the repository, it is the most recent, it is being developed.
Or [download v2.1 Beta](http://dl.programs74.ru/get.php?file=EspressifESP8266DevKitBeta) it is based on the current repository and soon it will be included in the SDK 1.5.3, the final version will be released after the test.

closed 2016-04-26 by CHERTS

chall3ng3r, 2016-05-10:
Thanks, gonna download the beta :)