cherts/esp8266-devkit #35: dht22_thingspeak resets

SoniaZotz, 2016-11-04:
dht22_thingspeak  

==== System info: ====
SDK version:2.0.0(5a875ba) rom 0
Time = 90302
Chip id = 0x2f842
CPU freq = 80 MHz
Flash size map = 
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 29016, room 16 
tail 8
chksum 0x7b
load 0x3ffe8000, len 2088, room 0 
tail 8
chksum 0xfa
load 0x3ffe8830, len 5864, room 0 
tail 8
chksum 0x01
csum 0x01
âäoäròo|ì$$$$ âr$j
==== System info: ====
SDK version:2.0.0(5a875ba) rom 0
Time = 89818
Chip id = 0x2f842
CPU freq = 80 MHz
Flash size map = 
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 29016, room 16 
tail 8
chksum 0x7b
load 0x3ffe8000, len 2088, room 0 
tail 8
chksum 0xfa
load 0x3ffe8830, len 5864, room 0 
tail 8
chksum 0x01
csum 0x01
âäoäròo|ìdlll`ã{Ûlj
==== System info: ====
SDK version:2.0.0(5a875ba) rom 0
Time = 90074
Chip id = 0x2f842
CPU freq = 80 MHz
Flash size map = 
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 29016, room 16 
tail 8
chksum 0x7b
load 0x3ffe8000, len 2088, room 0 
tail 8
chksum 0xfa
load 0x3ffe8830, len 5864, room 0 
tail 8
chksum 0x01
csum 0x01
âäoìÛ{ûn|äll$$ âr$j

valkuc, 2016-11-04:
To fix, remove code related to "Flash size map" output.

SoniaZotz, 2016-11-05:
Thank you valkuc

can you please explain why that was a problem

valkuc, 2016-11-05:
I have same issue in my own project long time ago. Either after adding boot loader support, either after removing it... Don't remember now.
Try to play with line `DHT22_DEBUG("Flash size map = %s\n", FlashSizeMap[system_get_flash_size_map()]);`
Modify it to `DHT22_DEBUG("Flash size map = %d\n", system_get_flash_size_map());` and see is app will still go into reboot loop, then try `DHT22_DEBUG("Flash size map = %s\n", FlashSizeMap[0]);`.
There can be only 2 causes of that:
 - either call to system_get_flash_size_map() cause fatal exception
 - either access to FlashSizeMap array cause fatal exception

CHERTS, 2016-11-05:
Fixed in commit https://github.com/CHERTS/esp8266-devkit/commit/350b751532d2b83cb6803660b16b275823d7657b

closed 2016-11-05 by CHERTS