cherts/esp8266-devkit #37: Advantage over the official devkit?

electronicsguy, 2016-12-07:
Hi,
   Great work. I had a simple question, sorry if it is too obvious or has been answered somewhere already - why should I be using this over the official SDK? Is there any specific advantage?

krzychb, 2016-12-07:
Esp8266-devkit kit contains official SDK.
More on user experience  - http://www.esp8266.com/viewtopic.php?f=9&t=820

electronicsguy, 2016-12-08:
@krzychb Yes I know that and I also went thru the discussion page (admittedly not all the 100s of pages). But what exactly can I do with this that I cannot with the official firmware?

krzychb, 2016-12-08:
Hi @electronicsguy,

Sometime ago I have been an looking for an ESP8266 toolchain for Windows platform. Searching for options I have been surprised to see that several people developed their own solutions to work with ESP8266 instead of using what is provided by Espressif (or maybe there was no Espressif's toolchain by that time?). At certain point I have been attracted by Mikhail Grigorev's esp8266-devkit. Everything else I have approached so far was for me to convoluted to apprehend, install and use. Checking esp8266-devkit I have been really amazed how quickly it was possible to install it, get up and running.

I think Espressif recognized this issue for ESP8266 and for the new chip ESP32 provided great [esp-idf](https://github.com/espressif/esp-idf) that is now my platform of choice for this new processor.

Krzysztof

electronicsguy, 2016-12-12:
@krzychb Thanks for the explanation Krzysztof. For a newbie who wants to start off in the present, what do you recommend? This toolchain or Espressif's official route?

closed 2017-02-22 by CHERTS