cherts/esp8266-devkit #39: Unable to link (`.text' will not fit in region `iram1_0_seg')

bsculley, 2017-02-21:
In the current version of the SDK (2.1.1) I am unable to link a project ([https://bitbucket.org/scargill/hackitt]) that previously worked.  

The error is: section `.text' will not fit in region `iram1_0_seg'.  

Any help in understanding what this means and how to troubleshoot it is appreciated. Especially understanding what items are included in the .text section and the relative space used by each item.

I'm running Windows 10 with Eclipse Neon as the IDE and the default eagle.app.v6.ld file located in C:\Espressif\ESP8266_SDK\ld.

CHERTS, 2017-02-22:
You need to use ESP module with a large amount of flash memory.

closed 2017-02-22 by CHERTS

bsculley, 2017-02-22:
This is not about memory size on a particular hardware module.

 

I have not even gotten to the point of flashing the module, I’m just trying to link the executable.

 

I am using the default eagle.app.v6.ld file:

 

/* This linker script generated from xt-genldscripts.tpp for LSP . */

/* Linker Script for ld -N */

MEMORY

{

  dport0_0_seg :                        org = 0x3FF00000, len = 0x10

  dram0_0_seg :                         org = 0x3FFE8000, len = 0x14000

  iram1_0_seg :                         org = 0x40100000, len = 0x8000

  irom0_0_seg :                         org = 0x40210000, len = 0x5C000

}

 

PHDRS

{

  dport0_0_phdr PT_LOAD;

  dram0_0_phdr PT_LOAD;

  dram0_0_bss_phdr PT_LOAD;

  iram1_0_phdr PT_LOAD;

  irom0_0_phdr PT_LOAD;

}

 

There is some issue here that needs to be resolved.  I’m willing to work on it but have no idea what to do.

 

Thank you.

Bob

valkuc, 2017-02-22:
The error is telling you that you don't have enough space in IRAM segment. If you previously was compiling against SDK 1.5.x, in new UDK SDK version 2.0.0 used as default and this version use more IRAM for it's needs. Check your code and annotate all non-ISR methods with ICACHE_FLASH_ATTR - if not help, google it. This issue is well known and it's not related to UDK.

bsculley, 2017-02-22:
Thanks for the response.

 

Just for clarity, and because I didn’t write the code I’m trying to compile, how can I distinguish an ISR method from any other?

 

Bob

valkuc, 2017-02-22:
Well... ISR handlers are ISR handlers... This require knowledge of firmware you trying to build. I can give you only one clue - check for gpio_intr_handler_register - this is GPIO interrupt register handler.

bsculley, 2017-02-22:
Ok, thanks. I think I can do that.

 

I appreciate your help.

 

Bob

valkuc, 2017-02-22:
Fast look at firmware you trying to build I have noticed that the methods already have ICACHE_FLASH_ATTR attribute specified. You can also try to build against old SDK - just change SDK_BASE in makefile to point to ESP8266_SDK_1541

bsculley, 2017-02-22:
Great, can you point me to where I can download the 1541 version of the SDK?

 

Bob

valkuc, 2017-02-22:
http://bbs.espressif.com/viewtopic.php?f=46&t=2198
http://bbs.espressif.com/viewtopic.php?f=46&t=2376

bsculley, 2017-02-22:
Success!!  The project builds without errors using this version of the SDK.

 

Thanks again,

Bob

 

http://bbs.espressif.com/viewtopic.php?f=46 <http://bbs.espressif.com/viewtopic.php?f=46&t=2198> &t=2198
http://bbs.espressif.com/viewtopic.php?f=46 <http://bbs.espressif.com/viewtopic.php?f=46&t=2376> &t=2376

bsculley, 2017-02-22:
Is it possible to know which code is included in the iram segment and the size of each module?

 

I would like to use the current SDK, but have no way to know by how much the limit is exceeded or which modules might be optimized or eliminated to fit.

 

Bob

bsculley, 2017-02-27:
Ok, optimisation – I’m sure of it…

 

The older versions of the SDK use up less room – actually 1.5 onwards is pretty much ok, but there’s still room left depending on the optimisation and the setting I think of that Eagle file.

 

 

From: Bob Sculley [mailto:bsculley@IntegratedSoftwareSvcs.com] 
Sent: 22 February 2017 14:11
To: 'CHERTS/esp8266-devkit' <reply@reply.github.com>
Cc: pete@scargill.org
Subject: RE: [CHERTS/esp8266-devkit] Unable to link (`.text' will not fit in region `iram1_0_seg') (#39)

 

Success!!  The project builds without errors using this version of the SDK.

 

Thanks again,

Bob